TheCrypt0/yi-hack-v4 #249: Support Yi Dome Guard

Sim7b, 2020-07-02:
Hello,

Did you plan to support Yi Dome Guard : https://www.aliexpress.com/i/4000674706324.html ?

Thanks

antonello69, 2020-07-04:
Buongiorno 
Anche io ho comprato una Yi Dome Guard ma sembra non si possa hackerare con questo firmware? 
Speriamo ci sia un aggiornamento
Grazie

flighter18, 2020-07-17:
Hallo 
Does anybody know more about this camera?
Is this really a YI camera?
I also have own but had no success with installing the firmware. 

Best regards

artemy, 2020-07-27:
Dome guard seems to be Allwinner-based https://github.com/roleoroleo/yi-hack-Allwinner

Zephael86, 2020-07-29:
@artemy : thank you, I have a look

antonello69, 2020-07-31:
Io sto provando il firmware Allwinner su dome guard ma non riesco a installarlo qualcuno è riuscito?

nicstoun, 2020-12-14:
> @artemy : thank you, I have a look

Hi, did you check try this firmware?