TheCrypt0/yi-hack-v4 #251: I bricked my camera!

tunerooster, 2020-07-06:
I bricked my camera (Yi Outdoor 1080p YHS-3017).  How can I recover?  I found and downloaded the original firmware "3.0.0.0D_201809111054home_h30m"  from YiTechnologies but when I install it on the sd card named "home_h30", does not work.  I am unable to find any instructions describing how to install this firmware on this camera.  Please HELP!!!

P.S.  accidentally deleted much of the root partition on the camera (oops).

TheCrypt0, 2020-07-06:
Hi @tunerooster,
Fortunately you should be able to recover the camera, download the [latest version of the firmware for your camera](https://github.com/TheCrypt0/yi-hack-v4/releases/download/v0.2.0/yi_hack_v4_v0.2.0__yi_outdoor.zip) extract it and put `home_h30` and `rootfs_h30` on the sd card.

Reboot the camera and you should be fine. 

If it doesn't work, first flash an older version of the firmware and then re-flash the latest one.

tunerooster, 2020-07-06:
That worked!  I had already tried installing v0.2.0, and it didn't work.  As you suggested, I installed v0.1.1 and it worked, followed by v0.2.0, which then worked!  I would never have guessed to try that...

Thank you very much!

Alex2x3x, 2020-07-14:
Does it have the timestamp embedded

bednim, 2020-07-26:
@TheCrypt0 hi, is it possible to update from 0.11 to 0.20 without sd card? im asking cause my camera is pretty high on the post...

thanks!

TheCrypt0, 2020-07-26:
Hi @bednim, unfortunately the firmware files must be on the SD card to be able to flash a new version, this is because how the bootloader handles the updates.