TheCrypt0/yi-hack-v4 #254: Camera support Victure SC220

the0s, 2020-07-25:
Will this firmware work with Victure Sc220 ? it seems to be a disguised yi camera.

timkrins, 2020-09-23:
Just FYI and anyone interested - The chipset of the Victure SC220 camera is a Realtek RTS3903N

the0s, 2020-09-24:
> Just for anyone interested - The chipset of the Victure SC220 camera is a Realtek RTS3903N

Hey TimKrins, 
do you know if this firmware hack will work with the victure?

timkrins, 2020-09-24:
I don't think so. See issue https://github.com/TheCrypt0/yi-hack-v4/issues/29

tarifari, 2022-06-11:
> > Just for anyone interested - The chipset of the Victure SC220 camera is a Realtek RTS3903N
> 
> Hey TimKrins, do you know if this firmware hack will work with the victure?

this one https://github.com/cjj25/Yi-RTS3903N-RTSPServer

dont know if you still need it, but rtmp is possible

mjpradeepraj, 2023-03-24:
> > > Just for anyone interested - The chipset of the Victure SC220 camera is a Realtek RTS3903N
> > 
> > 
> > Hey TimKrins, do you know if this firmware hack will work with the victure?
> 
> this one https://github.com/cjj25/Yi-RTS3903N-RTSPServer
> 
> dont know if you still need it, but rtmp is possible

I tried  https://github.com/cjj25/Yi-RTS3903N-RTSPServer but does not appear to be working. The camera keeps just rebooting. even the backup script has not run.