TheCrypt0/yi-hack-v4 #258: How you achieve the pem certificate from xiaomi binary key

mendellev82, 2020-08-03:
I am quite interest on this function generate_pem_certificate_from_xiaomi_binary_key in this script init_sysroot.sh
You did mention you spend quite some times on this function. Can you explain in more details how you achieve to write out this function? For example

1. How you know or determine the format/structure of the pub_key?
   For me it just a binary file, how we start to guess or determine its format?
   Since you divided it into modulus and exponents part then construct a der format, so initially it should not a standard der 
   format right because you concatenate the header and mid-header become a der format?
2. How you get the header and mid-header part?
3. How you know the der structure require by OpenSS pKey to convert to pem ? Play around with the OpenSSL generated one? 
4. Is that a must to use pem format to perform decrypt which cannot directly use the der format? 

Sorry I am a newbie so too much is over my scope. I hope you can point the right direction in order to save the time for me. Because it is time consuming without any idea on it to learn step by step.

TheCrypt0, 2020-08-05:
Hi @mendellev82, thank you for the great questions!

It's been a while since I wrote that function but in couple of days I gathered my old notes and remembered how it was done.

Basically, when an update is found, the camera executes some checks on it and calls a binary named `rsa_pub_dec` (or something really close) which decrypts the encrypted part of the update (see [`extract_fw_update()`](https://github.com/TheCrypt0/yi-hack-v4/blob/046078f87c62e097237251d33003aa070bb4c1d1/scripts/init_sysroot.sh#L160) for a better explanation on which part is encrypted and which is not).

I focused on the `rsa_pub_dec` binary and after some reverse engineering it was clear that it was using the `pub_key` file to do its job. No other files were used to get the public key and this mean that all needed was in that file. Remember, the minimum info you have to know to create a public key is the _modulus_ and _exponent_ which must be inside of that file.

Hoping the file wasn't encrypted itself I started by looking at its size, which was exactly `259` bytes. How convenient, this is exactly the sum of a 2048 bit modulus (256 bytes) and a 3 bytes exponent. After this discovery, all that was left to do it was to convert this binary file to something OpenSSL could use, what we are aiming at is the ASN.1 format. This is where [this great stackoverflow answer](https://stackoverflow.com/questions/27568570/how-to-convert-raw-modulus-exponent-to-rsa-public-key-pem-format) comes in. 

I had to make some guesses and a lot of tries before getting an usable public key, this was further complicated by the way the update file is constructed (again see the `extract_fw_update` function) and I wasn't sure that I was trying to decrypt the right portion of it. Fortunately after some time I ended up with a working script that could extract and decrypt the Xiaomi's provided update.

I couldn't remember why I ended up using PEM instead of DER but it could be because during testing I found it easier to edit it instead of a binary DER. I was exhausted but happy that it worked, and that was what mattered in that moment ;)

Hope I cleared any doubts you might had, if not feel free to ask!

mendellev82, 2020-08-08:
Thank you so much for your reply. Can you elaborate a bit how to perform reverse engineering on rsa_pub_dec? Using IDA or Hex editor? The info is quite limit base on the above two methods or do you have any more smart method?

TheCrypt0, 2020-08-08:
I often use IDA in combination with Ghidra. They are both very powerful tools.

mendellev82, 2020-08-09:
@TheCrypt0 Hi, thanks for your info. 

May I know What do you do if you find the file going to reverse engineering is obfuscated or encrypted for example what method or plugin you use in IDA or Ghidra to do the decryption? Can you just briefly explain the general steps? 

Second question is that the rsa_puv_dec is obfuscated/encrypted base on your previous experience?

TheCrypt0, 2020-08-10:
Both IDA and Ghidra can reverse engineer a whole bunch of executable formats, including ELF, but most importantly they are platform independent. You can statically analyze a Linux executable on a Windows machine.

The `rsa_pub_dec` is a normal executable, IIRC no obfuscation or encryption should be present.

mendellev82, 2020-08-10:
@TheCrypt0  Ok thank you

mendellev82, 2020-08-16:
@TheCrypt0 When I read back some of the previous issues, I found this part is interesting. 

[https://github.com/TheCrypt0/yi-hack-v4/issues/200#issuecomment-623561467](url)

You mention you are using virtual device to perform the RTSP. I don't know what is your viewd concept.
What I am curious is since you need a stable stream file (I don't know call it stream file is correct or not) for RTSP, 
Why you don't simple write the stream data to a file in /tmp and then stream it through RTSP and need to be so troublesomes using the virtual device? More confuse me is if not mistake FUSE is a filesystem. Why so trouble to purposely create a filesystem simply only for stream file. Any reason or benefit to using FUSE virtual device or it is a must?

Since in your extract firmware script you also using a virtual device for jffs2 image then why you don't simply use that method to create a virtual device instead of FUSE?

Hope you can clear my doubts to help me understand this embedded Linux better.
Thank you.