TheCrypt0/yi-hack-v4 #266: Yi Dome 1080p will not flash firmware

campanel54, 2020-08-24:
Followed instructions to the tee, i.e., FAT32 format of 16G micro SD, put "home_h20" and "roots_h20" at top directory... no hidden files.  Putting it in the camera and powering up it just boots as normal?  Creates a "log" and "record" directory and populates them with files.  No firmware flash?
Also tried firmware for Dome and Cloud Dome but they did the same thing.  Any ideas will be much appreciated.  Thanks in advance!
Mic

linusali, 2020-08-29:
I had the same issue with a YI Dome 1080p I got today from Amazon. I used firmware from https://github.com/roleoroleo/yi-hack-MStar#supported-cameras to get it working. Firmware I used was h201c_0.3.5.tgz. Hope it helps

MS

campanel54, 2020-08-29:
YES!  looks like it’s working!
Thank You for the reference!

Mic

dum-skywalker, 2021-05-22:
worked for me too after many tries with yi-hack 3, 4, 5 & all-winner-v2.
MStar has worked with my Dome Camera 1080p BFUS.

yonigold1, 2021-05-30:
Sorry quick question. How do you guys find out if your Dome Camera is BFUS etc? I have 2 Dome Cams and cant figure it out.