TheCrypt0/yi-hack-v4 #267: Fixed IP Address?

silent700, 2020-08-25:
Is there any way in the yi-hack firmware or web interface to set a fixed IP address?  DHCP-only is not very flexible.  I realized the stock firmware has the same limitation.

norrgrd, 2020-08-25:
You can probably just do a static DHCP mapping on your router.

silent700, 2020-08-25:
Sure that works but doesn't do much good if I want to pre-configure a camera and take it to another network, especially one that I don't control.

norrgrd, 2020-08-25:
That is very true, didn't think of that.

What about setting it with ifconfig?

To set the IP
ifconfig wlan0 IPADDRESS netmask 255.255.255.0 up

To set gateway
route add default gw IPADDRESS

To set DNS
echo "nameserver IPADRESS" > /etc/resolv.conf

Should do the trick!

norrgrd, 2020-09-01:
Did it work @silent700 ?

silent700, 2020-09-02:
I have not had a chance to hook the camera back up to test, but I believe I tried to set the IP info manually earlier and the ifconfig command was not present.  The OS appears to be pared down to the bare essentials.

norrgrd, 2020-09-02:
I know ifconfig works but I am not sure to which extent! I no longer have any camera running this firmware as it broke the rewind functionality for me through official app which was a deal breaker, therefore I cannot test either.

Johnr24, 2020-09-02:
I think this has worked for me!
I did get however

`/home/yi-hack-v4 # route add default gw 192.168.1.1
route: SIOCADDRT: File exists`

Johnr24, 2020-09-02:
I ran this 

'/home/yi-hack-v4 # ifconfig wlan0 192.168.1.214 netmask 255.255.255.0 up
/home/yi-hack-v4 # route add default gw 192.168.1.1
route: SIOCADDRT: File exists
/home/yi-hack-v4 # echo nameserver 192.168.1.1 /etc/resolv.conf
nameserver 192.168.1.1 /etc/resolv.conf
/home/yi-hack-v4 # cat /etc/resolv.conf
nameserver 192.168.1.1
/home/yi-hack-v4 #'

Will this persist after a reboot?

raffifu, 2022-03-06:
I have tried this method and works.
1. Create startup script in `/tmp/sd/yi-hack-v4/startup.sh`
2. Input command for set static ip address
```
#!/bin/sh

# set ip
ifconfig wlan0 192.168.43.41 netmask 255.255.255.0 up

# set gateway
route add default gw 192.168.43.1

# set dns
echo nameserver 192.168.43.1 /etc/resolv.conf
```