TheCrypt0/yi-hack-v4 #268: Rewinding in application

norrgrd, 2020-08-25:
Hello,

Has anybody else got rewinding to work? As soon as I try rewinding it just tries to connect to the camera forever until timeout.

They way I use the camera is with a 32 GB MicroSD-card with loop recording, I get notifications from the app and then I'd like to view them but this is not working at all. It's very tedious having to ftp to the camera every time to get the footage. The only way I can actually see anything else but live is by using the "include videos/photos in activity alert through secured cloud service" option enabled.

74ls04, 2020-09-04:
I believe the issue has to do with the timezone. If you set the timezone on the web interface it somehow messes up the app so I would try removing that and rebooting.

norrgrd, 2020-09-06:
Unfortunately this seems to do nothing. I've tried so many different combinations, setting timezone through TZ-file, app, combination of them etc but nothing seems to change it.

There's a different scenario everytime I reset the camera, for example right now I can playback everything that was recorded the first day but as soon as I go over 00:00 it won't load. If I start it at 23:59 (or 11:59 PM) and let it run past 00:00 it works...