TheCrypt0/yi-hack-v4 #269: startup.sh not present

adrianoasso, 2020-08-25:
Hi,

i have installed yi-hack-v4 on my "Yi-Cloud Dome Camera 1080p" for integrate it on homeassistant but, not working.
I have opened the system.sh script in /home/yi-hack-v4/script and at the end of file i noticed that the "startup.sh" script is not present in any folder.

It's normal?
Thanks
Adriano

TheCrypt0, 2020-08-27:
Hi Adriano,
If the installation isn't successful, I'd suggest you to download an older
version of the fw, flash it, and then re-flash the latest version. This
should fix any issue.

Let me know if it works!

>

adrianoasso, 2020-08-28:
Hi,
thanks for yout response.
The problem is still present. The script is not present in both paths:

/tmp/sd/yi-hack-v4/startup.sh -> isn't present "yi-hack-v4"
/home/hd1/yi-hack-v4/startup.sh -> isn't present "hd1"

Thanks.

TheCrypt0, 2020-09-28:
Hi again,
Oh now I understand! That's a way to run external scripts from the SD, it's to let user have their "startup.sh" without having to edit the one on the camera.

If you didn't create your own script on the SD it's normal you don't have it.

closed 2020-09-28 by TheCrypt0