TheCrypt0/yi-hack-v4 #271: Camera CMSXJ19E

xray11, 2020-08-26:
I have new 2K model CMSXJ19E.

Would this customized firmware also working with this model?
I´m not sure with the chipset.

croogie, 2020-09-02:
does the like mean that it will work? I don't want to brick my cam 😉

xray11, 2020-09-02:
I'm also asking on this, so no statement from my side. Hopefully there is someone more experienced... 

Or you can try it and then give us a feedback😀 (just a joke)

coendeurloo, 2021-01-21:
Anyone got some updates on it working on this recent model?

omracr, 2021-02-26:
That's what i need!!
An imilab integration for home Assistant waiting

cstrassburg, 2021-06-18:
not for home assistant integration but how to hack the camera and other details, you can look here: 
[cmsxj19e-hacks/](https://github.com/cstrassburg/cmsxj19e-hacks/)
I'm working now at documentation and scripts for installing RTSP, Motor control and Onvif
RTSP and Motor control are working.