TheCrypt0/yi-hack-v4 #272: Only yellow light is blinking

cooljackgi, 2020-08-29:
[./rmm][1/19/6:41:56:825]: (0), failed with 0xa0188005!
[./rmm][1/19/6:41:56:827]: HI_MPI_VI_GetFrame err, vi chn 0 a0078005
[./rmm][1/19/6:41:56:832]: HI_MPI_VI_GetFrame err, vi chn 0 a0078005
[./rmm][1/19/6:41:56:837]: HI_MPI_VI_GetFrame err, vi chn 0 a0078005
[./rmm][1/19/6:41:56:841]: HI_MPI_VI_GetFrame err, vi chn 0 a0078005
[./rmm][1/19/6:41:56:846]: HI_MPI_VI_GetFrame err, vi chn 0 a0078005
[./rmm][1/19/6:41:56:850]: HI_MPI_VI_GetFrame err, vi chn 0 a0078005
[./rmm][1/19/6:41:56:855]: (0), failed with 0xa0188005!
[./rmm][1/19/6:41:56:856]: HI_MPI_VI_GetFrame err, vi chn 0 a0078005
[./rmm][1/19/6:41:56:861]: HI_MPI_VI_GetFrame err, vi chn 0 a0078005
[./rmm][1/19/6:41:56:866]: HI_MPI_VI_GetFrame err, vi chn 0 a0078005
[./rmm][1/19/6:41:56:870]: HI_MPI_VI_GetFrame err, vi chn 0 a0078005
[./rmm][1/19/6:41:56:875]: HI_MPI_VI_GetFrame err, vi chn 0 a0078005
[./rmm][1/19/6:41:56:879]: HI_MPI_VI_GetFrame err, vi chn 0 a0078005
[./rmm][1/19/6:41:56:884]: (0), failed with 0xa0188005!
[./rmm][1/19/6:41:56:885]: HI_MPI_VI_GetFrame err, vi chn 0 a0078005
[./rmm][1/19/6:41:56:890]: HI_MPI_VI_GetFrame err, vi chn 0 a0078005
[./rmm][1/19/6:41:56:895]: HI_MPI_VI_GetFrame err, vi chn 0 a0078005
[./rmm][1/19/6:41:56:899]: HI_MPI_VI_GetFrame err, vi chn 0 a0078005
[./rmm][1/19/6:41:56:904]: HI_MPI_VI_GetFrame err, vi chn 0 a0078005
[./rmm][1/19/6:41:56:909]: HI_MPI_VI_GetFrame err, vi chn 0 a0078005
[./rmm][1/19/6:41:56:913]: (0), failed with 0xa0188005!
[./rmm][1/19/6:41:56:914]: HI_MPI_VI_GetFrame err, vi chn 0 a0078005
[./rmm][1/19/6:41:56:919]: HI_MPI_VI_GetFrame err, vi chn 0 a0078005
[./rmm][1/19/6:41:56:924]: HI_MPI_VI_GetFrame err, vi chn 0 a0078005
[./rmm][1/19/6:41:56:928]: HI_MPI_VI_GetFrame err, vi chn 0 a0078005
[./rmm][1/19/6:41:56:933]: HI_MPI_VI_GetFrame err, vi chn 0 a0078005
and
Filesystem           1K-blocks      Used Available Use% Mounted on
/dev/mtdblock5           12992      8232      4760  63% /home
Filesystem           1K-blocks      Used Available Use% Mounted on
/dev/root                 1280      1136       144  89% /

/home:
total 1
-rwxr-xr-x    1 root     root            22 Dec 14  2018 homever
drwxr-xr-x    8 root     root             0 Jan  1  1970 .
drwxr-xr-x   16 root     root             0 Jan  1  1970 ..
drwxr-xr-x    8 root     root             0 Jan  1  1970 app
drwxr-xr-x    4 root     root             0 Jan  1  1970 base
drwxr-xr-x    4 root     root             0 Dec 14  2018 hisiko
drwxr-xr-x    2 root     root             0 Apr 19  2017 lib
drwxr-xr-x    8 root     root             0 Jan  1  1970 yi-hack-v4

/home/app:
total 4094
-rwxr-xr-x    1 root     root       1871444 Dec 14  2018 rmm
-rwxr-xr-x    1 root     root       1009392 Dec 14  2018 oss
-rwxr-xr-x    1 root     root        575872 Dec 14  2018 cloudAPI_real
-rwxr-xr-x    1 root     root        306052 Dec 14  2018 p2p_tnp
-rwxr-xr-x    1 root     root         92000 Dec 14  2018 cloud
-rwxr-xr-x    1 root     root         82698 Apr 19  2017 local
-rwxr-xr-x    1 root     root         51068 Dec 14  2018 dispatch
-rwxr-xr-x    1 root     root         49696 Dec 14  2018 mp4record
-rwxr-xr-x    1 root     root         34868 Dec 14  2018 recbackup
-rwxr-xr-x    1 root     root         29300 Dec 14  2018 crypt_file
-rwxr-xr-x    1 root     root         19378 Dec 14  2018 watch_process
-rwxr-xr-x    1 root     root         15772 Dec 14  2018 log_server
-rwxr-xr-x    1 root     root         14608 Dec 14  2018 upgrade_firmware
-rwxr-xr-x    1 root     root         14128 Dec 14  2018 arp_test
-rwxr-xr-x    1 root     root         12344 Dec 14  2018 main.bmp
-rwxr-xr-x    1 root     root          3128 Dec 14  2018 sub.bmp
-rwxr--r--    1 root     root          2069 Jan  1  1970 init.sh
-rwxr-xr-x    1 root     root          1152 Dec 14  2018 audio_config
-rwxr-xr-x    1 root     root           438 Jan  1  1970 cloudAPI
-rwxr-xr-x    1 root     root           337 Dec 14  2018 wp_cmd
-rwxr-xr-x    1 root     root            22 Dec 14  2018 .appver
-rw-r--r--    1 root     root            14 Mar 23  2019 .camver
drwxr-xr-x    8 root     root             0 Jan  1  1970 .
drwxr-xr-x    8 root     root             0 Jan  1  1970 ..
[... 454 lines cut ...]
hi3518e_aio            28582  0 
acodec                  8218  0 
sensor_i2c              1966  0 
hi3518e_jpege          51897  0 
hi3518e_h264e         175633  0 
hi3518e_chnl           33603  0 
hi3518e_venc          222260  3 hi3518e_jpege,hi3518e_h264e
hi3518e_rc             72013  0 
ssp_ms41909             7375  2 
hi3518e_vpss          251733  1 
hi3518e_viu           258203  1 hi3518e_vpss
hi3518e_isp           112460  4 sensor_i2c,hi3518e_vpss,hi3518e_viu
hi3518e_vgs           129053  0 
hi3518e_region         67272  0 
hi3518e_tde           121919  1 hi3518e_region
hi3518e_sys            97009  2 hi3518e_vpss,hi3518e_viu
hi3518e_base           59381 18 hi3518e_adec,hi3518e_aenc,hi3518e_ao,hi3518e_ai,hi3518e_aio,acodec,hi3518e_jpege,hi3518e_h264e,hi3518e_chnl,hi3518e_venc,hi3518e_rc,hi3518e_vpss,hi3518e_viu,hi3518e_isp,hi3518e_vgs,hi3518e_region,hi3518e_tde,hi3518e_sys
hi_media                5801  9 hi_mipi,acodec,hi3518e_tde,hi3518e_base
mmz                    22815  9 hi_cipher,hi3518e_aenc,hi3518e_ao,hi3518e_jpege,hi3518e_h264e,hi3518e_viu,hi3518e_tde,hi3518e_sys,hi3518e_base
cpld_periph             6979  2 
mt7601Usta            694750  1 
hi_rtc                  6086  0 
exfat                  94437  0 

any idea what to do?

beastwithin78, 2020-09-13:
Hello,

I have the same problem with my Yi-cloud Dome Camera 1080P , it died suddenly without reason and now i have a yellow blinking led, i've tried to upload again the 0.2.0 with the usual method without success.

aladin2000, 2021-01-13:
same problem with mine  
Any clue ?

cooljackgi, 2021-01-14:
try to get a new one from amazon (;