TheCrypt0/yi-hack-v4 #277: Newest YI 1080p home camera compatibility

knightbyte, 2020-09-07:
Hello,
I've recently purchased some new YI 1080p Home cameras from [Amazon](https://www.amazon.com/gp/product/B01CW4AR9K) that now have a firmware version of 8.2.0.0A_202007211802.  The current version for the hack doesn't appear to be working on them.
Has anyone else ran into this and found a solution?
Thank you in advance.
- knightbyte

knut1er, 2020-09-08:
Hello,
Same issue for me..

jbandman, 2020-09-15:
Similar issue with a new YI 1080p Home camera bought from Amazon UK recently. Firmware version is as OP. Current version for hack does not appear to be working and neither is the current Allwinner hack 0.1.9. Camera appears to restart a second time a minute or so after being started; very shortly after this it is back and functional under its standard firmware, the hack has not taken.

chaddoncooper, 2020-09-15:
@jbandman It's worth looking at the Wiki pages to check the model exact model. I would imagine your's is the Allwinner CPU. The latest release extracted to the root of a FAT32 formatted 16GB card (32GB+ I read have issues?) worked for me. I actually updated the firmware using the app then used the hack. The change comes when you go to access it via the ip:8080 otherwise it works as stock. It doesn't take long for the hack to complete, a couple of short reboots - very close to the length of normal behaviour.

jbandman, 2020-09-15:
Chad, thanks for that, wasn't expecting quite such a quick response! ip:8080 got me into the yi-hack config page; took a few goes to see the network stream through VLC to test but that seems to be because I needed to insert the port into the rtsp URL, VLC couldn't connect otherwise. Thanks for your help