TheCrypt0/yi-hack-v4 #282: Purchased license not received

tmp0r, 2020-09-17:
Hi everybody,

I know that's not a technical problem but in discord I don't get a answer: I purchased a license last saturday. Till now I didn't received the license file. How do I know if my donation has been processed or how long does it take to get the license file.

Thanks in advance and best regards,
Daniel (discord: temp0r)

TheCrypt0, 2020-09-17:
Hi @tmp0r,
Thank you for the support! Others had the same issue as you, see this: https://github.com/TheCrypt0/yi-hack-v4/issues/190#issuecomment-588160959

Would you mind replying to the confirmation email? I'll handle the issue as soon as I can.

Cheers

tmp0r, 2020-09-17:
Hi @TheCrypt0,

thank you for your reply. I've replyed to your mail. I'm going to close this issue :-)

Thanks again and best regards,
Daniel

closed 2020-09-17 by tmp0r