TheCrypt0/yi-hack-v4 #287: New YI Home 1080P AI+ BFUSY44GEE with FW 9.0.19.02_202007141512 supported?

RestOp, 2020-09-30:
Hello,
New YI Home 1080P AI+ BFUSY44GEE with FW 9.0.19.02_202007141512
I just got this cam https://www.amazon.es/gp/product/B01MR82NFZ
On the back side of Cam is BFUSY44GEE. Date on box 2020.09
I made a backup of FW. Here it is: https://drive.google.com/drive/folders/1ifFIDG4G0v-aIKrE_4kcy3elXd5Bsae3
Can not understand if it is supported or not. Could you please look and say if I need to send it back to amazon?

Thanks a lot!

vigne, 2020-10-17:
Hello,

are there any news? I too have a Yi 1080p AI+ (BFUSY44TR) running the same firmware version and fail to install the hack. I downloaded the official version 8.2.0 but failed to flash that too, although I couldn't find any instructions on how to properly revert back to the previous firmware version.

Appreciate your work!

Cheers

RestOp, 2020-10-19:
> Hello,
>
> are there any news? I too have a Yi 1080p AI+ (BFUSY44TR) running the same firmware version and fail to install the hack. I downloaded the official version 8.2.0 but failed to flash that [...]

Please https://github.com/roleoroleo/yi-hack-Allwinner/issues/157