TheCrypt0/yi-hack-v4 #288: Camera not recording on SD when cloud deactivated & rtsp activated

jdieztec, 2020-10-09:
I'm trying all but I cannot get the camera recording on the SD when the cloud is off and the rtsp is on.

As soon as I turn on cloud, I've got recordings on sd, but it does not record anything when is off and the Recording without cloud is on.

jdieztec, 2020-10-16:
Anyone?

jgmrs, 2020-10-22:
Hi, I'm facing the same issue here. Waiting for a new release..

catweazel1509, 2020-10-27:
same here - no recording unless cloud active

MichlW, 2020-11-02:
Same here - I really need this feature when cloud is disabled.

Karrypto, 2020-11-04:
same here - no SD recording with Cloud features disabled, "Recording without Cloud" switch has no effect

randy3k, 2020-11-17:
It is the reason why I decided not to pay to unlock rtsp. It is basically useless if one cannot disable cloud and record to sd at the same time.

jdieztec, 2021-01-19:
No news here?