TheCrypt0/yi-hack-v4 #290: Yi Outdoor Camera YHS.3017 Do not flash

engineerme, 2020-10-11:
Hello all,

I bought some of these Yi Outdoor 1080 Cams for my construction site. Model No. YHS.3017

I wanted to use RTSP stream. But I can not flash it. The latest firmware ist installed.

It turns yellow and than blue light come ON.

Can some help me`?

[image]

cklimowicz, 2020-10-16:
I have the same problem, the camera does not read files uploaded to the SD card. 
Help needed

cklimowicz, 2020-10-18:
I managed to fix the problem.
First upload: v4 0.1.1
Then upload: v4 0.2.0

This is how it worked for me.
[image]

engineerme, 2020-10-18:
> I managed to fix the problem.
> First upload: v4 0.1.1
> Then upload: v4 0.2.0
> 
> This is how it worked for me.
> [image]

This worked for me. I also used an other software to format the SD Card in FAT32.
Thanks!

Btje1977, 2020-10-28:
What tool did you use to format en which type of SD card? I used the 0.1.1 and no succes.

gorbleu, 2020-11-18:
hi, same issue for me 0.2 or 0.11 same result .... anyone have an idea ?

jprafael, 2020-12-26:
fyi I manged to get it to work in the same way as reported:

1. Format the SD card with a 16gb (or less) FAT partition. Ensure your software does not use exFAT. I used the "Disks" utility from ubuntu.
2. Put the files from version v.0.1.1 in the SD card.
3. Put the SD card in the camera. Boot it, wait for it to flash. Confirm in the browser the hack is successful.
4. Shutdown camera.
5. Put the files from version v0.2.0 in the SD card.
6. Put the SD card in the camera. Boot it, wait for it to flash. Confirm in the browser the hack is successful.

timcrouzet, 2021-01-10:
Hi, 
Thanks for confirming it worked! Could you check your camera serial number? I am planning to buy a 4CUS and would like to be sure it will flash. 
thanks very much!
Tim

jprafael, 2021-01-11:
@timcrouzet this is the model i got: https://fccid.io/2AFIB-YHS3017