TheCrypt0/yi-hack-v4 #293: Is this project dead?

jNk89, 2020-10-16:
Is this project dead?

nbendall34, 2020-10-17:
No, there is just a lot of behind the scenes work

jNk89, 2020-10-20:
i think this is not true, i still waiting since a few days for the licence file after i spend some dollars and also there was no update over a year, but on the cams website i can read more than once "in progess"

TheCrypt0, 2020-10-20:
Hi @jNk89 please see this https://github.com/TheCrypt0/yi-hack-v4/issues/190#issuecomment-588160959

mr-brunes, 2020-10-29:
I've opened the discord site and typed donate but nothing happens. Seems dead to me ...

alienatedsec, 2020-11-24:
> No, there is just a lot of behind the scenes work

I still have no empathy the way the project collects the money and is not issuing anything more in return. I understand your view and commitments to personal lives and other projects but this seems to be unfair.