TheCrypt0/yi-hack-v4 #294: YI Outdoor Camera 1080p 9CUS supported by this Custom Firmware

huntz, 2020-10-20:
Hi,
I'm currently running the yi-hack-v4 on YI Outdoor Camera 1080p 9CUS model.

It's possible to add the 9CUS model to the main README.md? It will be helpful for the new users.

Thanks!

JakubBialoskorski, 2020-11-03:
I was wondering if 9CUS are supported - good to know. Thanks @huntz !

AhleWurscht75, 2020-12-02:
> Hi,
> I'm currently running the yi-hack-v4 on YI Outdoor Camera 1080p 9CUS model.
> 
> It's possible to add the 9CUS model to the main README.md? It will be helpful for the new users.
> 
> Thanks!

Hi, i dont get the Rstp Stream running in vlc with this model. Can you help me?

nicmela, 2020-12-06:
Is possibile to use this hack to connect the 9CUS model to a synology via rtsp?
I would like to try but i'm a bit scared/concerned since this project seems pretty much dead according to discord...

timcrouzet, 2021-01-10:
Hi, do you know if the 4CUS is supported as well ?
Thanks
Tim

zsdd89, 2021-05-06:
> Is possibile to use this hack to connect the 9CUS model to a synology via rtsp?
> I would like to try but i'm a bit scared/concerned since this project seems pretty much dead according to discord...
> 
> Can you tell us how do you managed to let it work?

I have 9CUS version too (model YHS 3017) and only this yi-hack-v4 is working stable with my synology nas, i tried yi-hack-v5 from alienatedsec but there rtsp stream is very choppy, whole minutes are missing every hour from the recordings, it`s very unstable on this model