TheCrypt0/yi-hack-v4 #297: Yi home 1080p 6FUS can't upgrade firmware

rithgithub, 2020-10-30:
Yi home 1080p camera 6FUS load only yellow light. can't upgrade firmware. Have any way to upgrade firmware please help to provide step by step. Thank you

dls11, 2020-11-25:
I have the same problem with the BFUS model. The yellow light is not flashing. There is no upgrade.

Kaszys, 2020-12-09:
Same problem here with the BFUS model. Actual firmware is 9.0.19.02_202007141512.

cabrauck, 2020-12-12:
Same here with BFUS

swayer89, 2020-12-23:
The camera is not flashing the firmware

nico81, 2021-01-13:
Same here, fw 9.0.19.02_202007141512

guim31, 2021-01-13:
Same here with fw 9.0.19.02_202007141512 

Hope some fix is coming...!

sergiocasero, 2021-02-04:
Same error here, someone found any solution?

kicker83, 2021-02-04:
I found the solution using this firmware. https://github.com/roleoroleo/yi-hack-Allwinner-v2

sergiocasero, 2021-02-04:
@kicker83 you saved my day!! Thanks you so much! ;)