TheCrypt0/yi-hack-v4 #306: 1080p dome with 45US s/n

Zippeyrude, 2020-11-21:
I see that other repos have specific serial number compatiblity, but I cannot see 45US listed.

Does this repo support this version of the dome?

Also, im looking for nvr compatibility, presumably onvif, was wondering how close this repo was from including this?

Many thanks for hard work, appreciated!

bartlomiejborzucki, 2020-12-09:
It is the newest version of the firmware works with your 1080p dome 
https://github.com/Keroi/yi-hack-v4/releases

Zippeyrude, 2020-12-10:
Thanks
I'm hoping for onvif, the rtsp I can consume in vlc for example but nvr doesnt seem to accept unfortunately

The last release was May 20, is this repo still active m

bartlomiejborzucki, 2020-12-10:
I don't know. Fingers crossed for the new release.