TheCrypt0/yi-hack-v4 #308: Is this project already dead?

alienatedsec, 2020-11-24:
There are no updates since last year (May 2019), and other similar projects have their updates regularly. Unfortunately, it is all related to the type of chipset; however, this project promised ONVIF from day one, and I genuinely supported it with my two cameras and wanted to follow with the rest. I realised in time before throwing my money into the hole.

I am sure the licensing portal and their system are working perfectly, so the money collection is the only thing you can expect from this project.

Rushmed, 2020-12-01:
I am interested too.

influenist, 2020-12-03:
v5 anyone ... ? :-)

ljmerza, 2020-12-18:
he responds when it comes to someone donating money but for this? nah.

swayer89, 2020-12-23:
I am interested too.

jdieztec, 2021-02-15:
Really fast to answer to receive payments, no answer on any issue.

I think that if you have abandoned this you should have the decency to warn the new users before throwing their money on a dead project.

alienatedsec, 2021-02-23:
Please see my fork - there is still work to do on space and other areas, but it is in progress

TheCrypt0, 2021-02-23:
@alienatedsec Would you mind dropping me a pm on discord or a mail? I tried to find you on the server but without much luck.

Lately I've been testing some other forks with PTZ functionality and stuff and, if you are willing to, I'm sure we can join our forces to include a lot of fixes, features. I'll be more than happy to give you a jumpstart.

Let me know,
Crypto

alienatedsec, 2021-02-23:
@TheCrypt0 

[image]

alienatedsec, 2021-02-24:
> Lately I've been testing some other forks with PTZ functionality and stuff and, if you are willing to, I'm sure we can join our forces to include a lot of fixes, features. I'll be more than happy to give you a jumpstart.

@TheCrypt0 As much as your message had good intentions, and despite I have contacted you via Discord, it would be useful to continue our conversation and see how you want to contribute further.

Unfortunately, your RTSP donation system you created around your repository is no longer valid in my fork. This will mean that we could work together and 'join our forces' as you said, or we will explore our developments separate from each other.

I am happy to work either way.

closed 2021-02-24 by TheCrypt0

joseantgv, 2021-11-17:
> @TheCrypt0
> 
> [image]

How can I join the Discord server? Thanks.