TheCrypt0/yi-hack-v4 #313: Problem update remote version

SirkeS, 2020-12-08:
I have the same problem on my four cameras

/home/yi-hack-v4 # wget https://raw.githubusercontent.com/TheCrypt0/yi-hack-v4/master/VERSION
Connecting to raw.githubusercontent.com (151.101.64.133:443)
wget: note: TLS certificate validation not implemented
wget: TLS error from peer (alert code 40): handshake failure
wget: error getting response: Connection reset by peer

kani85, 2020-12-26:
Yes, it happened to me also.
In order to solve I connected via ssh and commented two lines at the end of script/system.sh referring to check_update.sh
I think this maybe is related to the version of wget inside busybox?

vcasado, 2020-12-27:
> Yes, it happened to me also.
> In order to solve I connected via ssh and commented two lines at the end of script/system.sh referring to check_update.sh
> I think this maybe is related to the version of wget inside busybox?

Thanks for the workaround, seems that it works!

TheCrypt0, 2020-12-28:
Hi guys, thanks for the report!
You are right, it seems like GitHub changed its minimum TLS scheme and the current `wget` on the camera doesn't have it supported.

Removing the lines you referred to along with the `cronjob` that calls `check_update.sh` (old version of the fe) will do the trick.

Hammerfest, 2021-01-02:
Can someone write a guide on how to do this for those who dont know how?

vcasado, 2021-01-02:
> Can someone write a guide on how to do this for those who dont know how?
> 
> (for example, I can ssh to the unit, I can ls/cd around, but the rest is lost on me)

This is what I did:

1. Log in via SSH.
2. Open file **system.sh** under folder **script**:  _vi /home/yi-hack-v4/script/system.sh_ 
3. Enter edit mode pressing key **'i'**.
4. Scroll down and comment the lines **83** and **85**:

    #$YI_HACK_PREFIX/script/check_update.sh
    #crond -c $YI_HACK_PREFIX/etc/crontabs

5. Press '**ESC**' to exit edit mode and then type "**:w**" and press Enter to save changes.
6. Reboot!

SirkeS, 2021-01-02:
Method 2:

Log in via SSH.
Open file ssystem.conf under folder script: vi /home/yi-hack-v4/etc/system.conf

Enter edit mode pressing key 'i'.

Scroll down and change value
CHECK_UPDATES=no

Press 'ESC' to exit edit mode and then type ":w" and press Enter to save changes.

Reboot!

Hammerfest, 2021-01-02:
Both of you mention OPEN, but not HOW to open, probably the only thing I am missing here.