TheCrypt0/yi-hack-v4 #314: Error installing the firmware

jplajpla23, 2020-12-09:
Hi I buy two Yi outdoor cameras and try install the Yi hack v4 in both cameras. The first I follow the steps and all ok, can acces to web interface the FTP and others functions.
But the second camera, I follow the steps and the camera is on(works on Yi app for windows the real time) and if I plug the SD card on computer sd show the folders trees, but if I try access to ftp or ssh or web interface not allow. I scan the network for show the open ports and the IP not show any port open.  
What can be wrong?

aleixsb, 2021-01-08:
I have the same problem. I have a Yi IOT Outdoor camera, but when I plug in the SD card it just boots as usual and doesn't flash anything. On the SD card I can see some logs written by the camera and some videos..

any idea?

nico81, 2021-01-13:
Same here for YI Home 1080p

eagle-phoenix, 2021-01-16:
me too, with Yi 1080p Home Camera (6FUSY213M5)

AngeloAna, 2021-01-27:
I have the same issue with Yi 1080p home cameras. The firmware will not load.

AngeloAna, 2021-01-27:
Never mind. There are 4 projects out there. You need to use the right one depending on camera model and current camera firmware version.

jplajpla23, 2021-01-27:
I finally can install firmware.
Put the camera in reset again and put the files again in se card

closed 2021-01-27 by jplajpla23

aleixsb, 2021-01-27:
Hu @jplajpla23 could you explain what you did to get the camera working with the firmware?

jplajpla23, 2021-01-28:
-Format again the SD card
-Put the files again in sd card
-Reset button on Camera
-insert the sd card