TheCrypt0/yi-hack-v4 #315: 60.000 DNS queries in 24 per Yi Outdoor Cam

waywit, 2020-12-20:
Hi, I am using 3 Yi "hacked" 1080p Outdoor Cams and they are working great in my HomeKit... but I also use a pinhole for adblocking in my home network... and all 3 Yi Cams are sending more than 60.000 DNS queries to my pinhole per day... which makes 87% of all DNS queries in my network. They are trying first to reach out to "raw.githubusercontent.com" and then to "github.map.fastly.net", each 30.000 times a day.

I have just blocked all 3 Cams to for these domains, which results in 30.000 queries a day, because they are now trying to reach "raw.githubusercontent.com" only, which will get blocked... After blocking everything is working correctly, including the cloud functionality over the Yi Cam App...

Why are they trying to reach GitHub in such a fast frequency? Is this related to the hack or to the cloud functionalities? Can this be stopped in some way?

Thanks and best regards :-)

weird81, 2020-12-27:
I have the same issue with Yi dome camera. Plus there is constant ~6% (whatever is the unit) Load Average of the camera's CPU. I believe it started in the last few weeks, maybe months. I don't recall seeing that before (but I can be wrong with that, last statement)

johnyeros, 2020-12-30:
same problem i just notice today. project is dead. the guy is quick to response for donation but he isn't doing anything else. It's sad.

TheCrypt0, 2020-12-30:
Hi, I replied on the same issue here: https://github.com/TheCrypt0/yi-hack-v4/issues/313#issuecomment-751606964

waywit, 2020-12-30:
Thanks for the workaround it is working fine... :-) But actually there will be no update check anymore... Hope you will find some time to setup a new release, as announced on discord...