TheCrypt0/yi-hack-v4 #316: Unable to Share Camera via YiHome app

eximo84, 2020-12-21:
I have two cameras both with YiHack v4 firmware. I still use the yihome app and recently noticed the cameras disappeared after I changed iPhone. 

I have reset and repaired the cameras and can use them however other iOS devices cannot see them if I sign in with my account or if I try and share them with another account I get invalid region error. 

Could this firmware be stopping this?

previously cameras appeared on all iOS devices with the same account.

Florian-Schmidt, 2020-12-22:
I think I was able to see my camera also after installing YI-HACK. Suddenly they disappeared but is still available via local network.

@eximo84 You already checked "private mode" under "Configurations -> General"?

eximo84, 2020-12-23:
> I think I was able to see my camera also after installing YI-HACK. Suddenly they disappeared but is still available via local network.
> 
> @eximo84 You already checked "private mode" under "Configurations -> General"?

Yes private mode is disabled. The app works for just 1 device fine. Just can’t get the cameras to share or if I login on another device with the same account they don’t appear.

eximo84, 2020-12-30:
Restored camera back to original firmware. Shared them fine.  Then installed latest version of YiHackv4 and sharing remains in tact.  Odd.

closed 2020-12-30 by eximo84

Florian-Schmidt, 2021-03-18:
@eximo84 Sorry, took some time as my microSD really had been broken. Received new card tand switched to project [Yi hack M-Star](https://github.com/roleoroleo/yi-hack-MStar). This project seems to be still active, camera is reachable again and connected to Yi app again.