TheCrypt0/yi-hack-v4 #319: Yi Outdoor - Stock firmware

LeidenSpain, 2021-01-12:
I have a Yi Outdoor with the hack installed, and it has broken.
I would need the original firmware to get it repaired, but I can't find it.

Do you have the mega link that is indicated, to be able to download it?

db25s, 2021-01-18:
interested also

LeidenSpain, 2021-01-18:
Hi,
the seller has sent me the original firmware

_Please find the firmware file attached and follow the instructions: 

HOW TO UPDATE? Please do not open or rename the file! Just copy it to your SD card.
How to install a firmware:
1) copy  the file to your SD card (use a card reader)
2) insert the memory card into device
3) turn on the device
4) the device will start updating by itself
5) the device is automatically updating with a constant yellow light until it completes and restarts again
6) when you hear the “Waiting to connect " voice alert again with a flashing yellow light indicating the user to connect the camera, continue syncing the camera with the app._

[home_h30m.zip](https://github.com/TheCrypt0/yi-hack-v4/files/5832407/home_h30m.zip)

luka-diplomski, 2021-03-19:
i bricked my outdoor also. 
this recovery doesnt help https://app.box.com/s/cibs7n1mgvhqaqjlidtveegu1uajt5yr/folder/23494084746
...nor does v0.1.1

I need rootfs for the new fw.
Please help