TheCrypt0/yi-hack-v4 #323: Fails after reboot

shhkid, 2021-02-05:
Hi,
when I try to reboot my 1080p outdoor cam via SSH, the camera does not reboot properly.
It does not get an IP address assigned, so there is no way to access it in this state.

Only a hard reboot (interrupting the power supply) let's the camera work again.
I tried to investigate the issue but couldn't find log files on the file system.
Can anyone confirm that issue?

Regards

coolerbaier, 2021-03-01:
Please provide Firmware Version which was on the CAM originally, HW-Version and procedure how you updated the FW-Hack

shhkid, 2021-03-01:
Hi,
CAM FW Version: 3.0.0.0D_201809111054
I updated to the hack version using the steps as described in the tutorial. 
I have this version: https://store.kamihome.com/products/yi-outdoor-security-camera. Where can I find a more specific HW-Version if there's one?

coolerbaier, 2021-03-01:
HW Version is on the back Model Number - mine is YHS.3020. I am with FW version 9.0.26.05_202011201756 and I cannot upload the hack - the CAM does not upload anything :-( so I am at stock-FW without access.