TheCrypt0/yi-hack-v4 #327: Yi Outdoor 1080 no SD Firmware update possible

coolerbaier, 2021-02-25:
Upgrade of the cam with SD card without success. 

Cam seems to ignore SD card files home_h30 and rootfs_h30 and starting normal with WiFi connection. FW-Version remaining 9.0.26.05_202011201756 in the App and in the SD card log after normal start. CAM is working fine with the yi home App, but no IP access with http, ssl, telnet.... Ping is working.

What I did already:
SD format in FAT 32, only the 2 files in SD-root copied with Linux, reset button before card insert, reset button with inserted card, various sd cards from 2 .. 15GB, Stock FW from #319 (home_h30m) 3.0.0.0A_201805151439, Stock FW renamed to home_h30, Stock FW zip-file on SD-root, -- all with no effect.

According Project page all versions of yi outdoor are supported with the hack files (home_h30 and rootfs_h30). Is it possible that 9.0x firmware needs other files or can not boot with SD files? However, when inserting a SD into running YI CAM it reboots and searches for files? 
Workaround available?

Anybody with same experience?

ChennyBaBy, 2021-02-25:
yeah i am not able to flash the outdoor cam with the latest release either, help would be much appriciated.

ChennyBaBy, 2021-02-25:
@coolerbaier hey man, i manage to get the firmware flashing, its not done yet but i thought to give you a shout, hope this helps u

Follow this issue here:
https://github.com/TheCrypt0/yi-hack-v4/issues/290

It looks like your have to format your sd card to be 4gb then install a earlier version first of 0.1.1

coolerbaier, 2021-02-25:
Tnx ChennyBaBy. 
I tried version yi_hack_v4_v0.1.1__yi_outdoor.zip and prepared the 4GB SD with 
sudo mkfs.vfat -F 32 -n 'yi' /dev/sda1
sudo mount /dev/sda1 /mnt/usb
then copy both files into the root of the SD card. 

Still no flashing. 
1. If CAM was already connected to the yi home app before insert SD card, it connects to the WLAN after some seconds blinking orange with a solid blue LED. YI home App shows still the FW 9.0.26.05
2. If CAM was reset by the key before insert the SD card, the "waiting to connect" voice comes after the "welcome to yi home camera" with blinking orange 

Might be a problem with this FW 9.0.26.05. Model is: YHS.3020 yi  Outdoor Camera 1080p; Date on the box shows 2021.01 

However, please try at your own and provide your results.

ChennyBaBy, 2021-02-26:
I would try doing the formatting on a windows machine. Or maybe try a
different as card

coolerbaier, 2021-02-26:
sure, I did formatting already on Windows with FAT32 and also FAT. Also formatting SD card in CAM with the yi home APP - no success.

fgekko, 2021-02-26:
I have the same issue with this FW 9.0.26.05. Model is: YHS.3020 yi Outdoor Camera 1080p

DennisUsesRPA, 2021-03-02:
I'll follow this issue since I'm experiencing the same.

bit-user, 2021-03-02:
I am experiencing the same.

Is this project still alive?

Mizerka, 2021-03-06:
same issue on YS2016 (dome 1080p) new out of box. stays orange for 10s and proceeds to boot without flashing

DennisUsesRPA, 2021-03-06:
So I managed to hack it :)

Instead of the Yi-Hack v4, I used an other way of hacking the Yi camera. For the IFUS-series, you can use the following: 
Navigate to https://github.com/roleoroleo/yi-hack-Allwinner-v2 where you want to pick the release: y21ga_0.1.4 (https://github.com/roleoroleo/yi-hack-Allwinner-v2/releases).

They also provide different suffixes along the way of other serial no's.

coolerbaier, 2021-03-07:
Thank you Denis . This worked for me 👍
My solution steps were:
1. Format the SD with the Yi-home App. (probably also with Windows possible)
2. Copy the extracted files from https://github.com/roleoroleo/yi-hack-Allwinner-v2/releases/download/0.1.4/y21ga_0.1.4.tgz to the root of SD card
3. Insert SD in CAM
4. Apply Power to the CAM
That`s it.
My FW 9.0.26.05. Model is: YHS.3020 yi Outdoor Camera 1080p; Date on the box shows 2021.01. y21ga works also for this setup.

closed 2021-03-07 by coolerbaier

Jeoffreybauvin, 2021-03-26:
That's weird, it's not working for me...

xvolte, 2021-08-01:
@coolerbaier @Jeoffreybauvin 
Same here, does not work.
Did you manage to do it again ?
is there a particular reason of using an old version ? (0.1.4?)

Thanks in advance

xvolte, 2021-08-01:
In fact, IT DOES WORK, i used latest y21ga version for my Yi outdoor 1080 cam YHS.3020.
The reason why i thought this didn't work is because of the cam that does not seem to be flashing anything, but it's so quick ... it's working like a charm.
Note : As i tested multiple times, maybe you need first to install the oldest 0.1.4, and then the latest... i don't know.

I'm running version 0.1.8 now.

THANKS

goatmoag, 2022-07-09:
2.2 worked fine. Keep in mind you have to leave the card in with the hack.