TheCrypt0/yi-hack-v4 #328: TO DO (suggest )  - Reboot schedule

nelutzu777, 2021-02-26:
I suggest implementing a Reboot schedule. 
Why: I use GDriveUploader, I noticed that the cameras freeze and no longer load on the GDrive. After rebooting cameras via telnet or web, everything returns to normal, old folders removed from Drive and uploaded new ones. I use YI Home 47US and YI Home 48US. On both models they periodically encounter this problem (Once a week for sure). I assume it has to do with loss  internet connection for couple seconds , respectively the GDRIVE script catches an error and stops. All the IP cameras I have met, even routers, have "service reboot time" in the settings and you can choose "daily", "weekly" or "monthly". Thanck you

dengke80622, 2021-11-19:
same issue with yi_home_1080p, system freezes and no respond with ping from other device.
Tried to schedule remote ssh command from openwrt but failed with shell file. output info shows below
----------------------------------------
#!/bin/sh
sshpass -p root ssh root@192.168.10.207 '/home/base/tools/reboot'
----------------------------------------
$ sh /etc/yi-cam_rbt.sh
killall: watch_process: no process killed
killall: mp4record: no process killed
killall: recbackup: no process killed
/home/app/localbin/reboot_by_watchdog: can't load library 'librt.so.0'
umount: can't unmount /tmp/cifs: No such file or directory
--------------------------------------
command below works if is used on yi-cam ssh terminal directly.
-----------------------------
/home/yi-hack-v4 # /home/base/tools/reboot
killall: watch_process: no process killed
killall: mp4record: no process killed
killall: recbackup: no process killed
umount: can't unmount /tmp/cifs: No such file or directory
------------------------------

Any suggestions for remote reboot command?

Thanks

nelutzu777, 2022-10-27:
After some research, here's how I solved it:
1. connect via TELNET or SSH to the camera
_crontab -e_
Press "_o_" (Letter "O" - Oliver, to add new line VI editor)
_0 3 * * * reboot now_     (reboot at 3:00AM daily)
Press Esc, type _:wq_ and press enter to exit vi
reboot

I use JuiceSSH for Android, so others may have other commands to save and edit

closed 2022-10-27 by nelutzu777