TheCrypt0/yi-hack-v4 #329: Yi Outdoor 1080p super high CPU load

ChennyBaBy, 2021-02-26:
I have the rtsp running but its very unstable, and i can only access the feed for about 30 secs before it dies, the CPU load of the camera is super high I have turned off everything i can deisable cloud, disable ftp, disable ssh, ntpd, and using the SD instead of HD for the rtsp stream but it's still not helping.

Does anyone know why? I've ssh in to  `top` and see what is happening.

[image]

coolerbaier, 2021-02-27:
ChennyBaBy, seems that you managed your CAM hacked now. Please tell me how? Which HW/SW Versions were on your CAM ?

DennisUsesRPA, 2021-03-02:
Follow

coolerbaier, 2021-03-07:
ChennyBaBy, I have the same CPU load with ./rmm Process. 
Btw, meanwhile I was successfull hacking my Yi Outdoor with yi-hack-Allwinner-v2 which is more up-to-date.

X3fxKupaA6jaHX83Whnn3xcYWnUCYhmjr2e2A8T, 2021-03-13:
> ChennyBaBy, I have the same CPU load with ./rmm Process.
> Btw, meanwhile I was successfull hacking my Yi Outdoor with yi-hack-Allwinner-v2 which is more up-to-date.

What did you have to do to get the Allwinner-v2 to install correctly?  
Appreciate any information you can provide.

This is being discussed here as well:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/97

coolerbaier, 2021-03-14:
I documented the solution which worked for me in #327