TheCrypt0/yi-hack-v4 #335: YI camera outdoor  1080 hack

nonome1, 2021-04-01:
Hello, I have a My FW 9.0.26.05 model camera.  Model is: YHS.3020 yi Outdoor Camera 1080p;  Date on the box shows 2021.01.  I can't manage the hack to be able to integrate it into home assistant.
1: I formatted my 32g micro card with the name ROOT and I put the 2 files at the base of the micro sd.
2: I insert my SD card into the camera then I start the camera.  it lights up orange fix then flashes blue and afterwards unable to connect with ip from the camera that I recovered in my router.

the hack does not work.  Thanks for your help

coolerbaier, 2021-04-19:
I documented the solution which worked for me in #327

Johanson2020, 2021-06-21:
Hello, 
I am in despair & need your help please.
I have the same problem, that the camera ignore the mod. firmware on the SD-Card
Firmware: 9.0.26.07_202104231123
FCC ID: 2AFIB-YHS3020

what can I do to narrow down the problem?
Sorry but i have no idea...

I tried:
YI Hack v4
YI Hack v5
Allwinner V2

Thanks a lot for your support!

https://github.com/alienatedsec/yi-hack-v5/issues/76