TheCrypt0/yi-hack-v4 #343: wifi disconnect - wpa may crashed, now reset

k3s4, 2021-05-14:
Hello,

I have issue with constantly disconnecting WiFi from my Yi Outdoor camera - 

`[./rmm][5/13/22:52:26:887]: ai_get_aacstream: get aenc stream select time out
[./rmm][5/13/22:52:29:71]: ai_get_aacstream: get aenc stream select time out
[./dispatch][5/13/22:52:29:792]: wifi disconnected, now reconnect wifi
[./rmm][5/13/22:52:30:718]: motrk img is too old. storetm:1620946349-725901, diff:1009
[./mp4record][5/13/22:52:33:144]: msg snd success[./dispatch][5/13/22:52:33:145]: got rcd heartbeat
[./watch_process][5/13/22:52:34:512]: cloud crashed![./rmm][5/13/22:52:41:538]: got a new motion stop. send_motion_stop_msg
[./rmm][5/13/22:52:41:561]: msg snd success[./dispatch][5/13/22:52:42:632]: wpa may crashed, now reset
[./dispatch][5/13/22:52:42:766]: no usb dev
`

`
          Mode:Managed  Frequency=2.447 GHz  Access Point: 20:76:93:4E:A5:32
          Bit Rate=135 Mb/s
          RTS thr:off   Fragment thr:off
          Encryption key:D2F5-928A-F48D-41FA-1334-E9DD-6D08-C0AC   Security mode:restricted   Security mode:open
          Link Quality=100/100  Signal level:-54 dBm  Noise level:-85 dBm
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0
`

after reboot it work only for couple of hours..
anyone has similar issue and fixed that somehow?

leviatan6, 2021-07-22:
I have the same problem.

k3s4, 2021-07-22:
@leviatan6 workaround - https://github.com/alienatedsec/yi-hack-v5/issues/54