TheCrypt0/yi-hack-v4 #355: RTSP unlock = scam?

ThatCoffeeGuy, 2022-01-12:
The donate section on discord is full of sketchy messages. The entire discord is overall full of Steam scam links, completely unmoderated. There are also people complaining about not receiving anything after donating.

codeclinic, 2022-01-18:
There's also 244 open issues. Make of that what you will. ;-)

ThatCoffeeGuy, 2022-01-24:
Thank you! I found it already, looks much better, although I've had some weird issue with it, but after some manual tweaking, it's somewhat working now.

jcdauchy, 2023-01-22:
> There's also 244 open issues. Make of that what you will. ;-)
> 
> You might want to take a look at this project: https://github.com/alienatedsec/yi-hack-v5

Can I flash the v5 firmware after using hacked v4 firmware ? Or I need to go back to stock firmware first ?

Thanks