TheCrypt0/yi-hack-v4 #356: Yi camera 1080 doesnt recognize firmware

TheAlchemist92, 2022-01-26:
I've tried to upgrade my camera with the new firmware but it doesnt flash.
I formatted a 1Gb SD in FAT32 and added the twoo files. when I power up the camera it turns on normally as if it does not recognize the files, and starts recording on the SD

pashok2398, 2022-04-17:
Same here.
Trying Outdoor 1080 nothing happened, i must tell that about year ago i did the same and everything worked.
Maybe its the new version of the Camera original firmware?

iwanttobefreak, 2022-04-18:
Same issue
My firmware version: 9.0.26.05_202011201756

okynos, 2023-01-12:
Same here, something to do here?

SavenNeer, 2023-01-31:
Same Here at YI Home 1080p

Digitalpeer1, 2023-03-22:
same as the guy above

okynos, 2023-03-23:
@SavenNeer @Digitalpeer1 Take a look at this repository https://github.com/roleoroleo/yi-hack-Allwinner-v2 checkout your Serial number and compatibility, there are different variants of yi-hack

SavenNeer, 2023-03-23:
> @SavenNeer @Digitalpeer1 Take a look at this repository https://github.com/roleoroleo/yi-hack-Allwinner-v2 checkout your Serial number and compatibility, there are different variants of yi-hack

yep, i had tried it and it works! Thanks!

Digitalpeer1, 2023-03-23:
ok thanks