TheCrypt0/yi-hack-v4 #364: How to remove v4 hack? (Original YI Home, 47US)

HaraldBB, 2022-09-04:
As "Unlocking RTSP after donation" is now a scam with no doubt, tried to move to v5. 
Unfortunately v4 appears to be somehow permanent now. While v5 is flashing, camera is connecting to wifi, but old v4 menu on web UI stays, but it's inactive. No telnet, ssh, ftp etc available.
Tried to use built-in reset function by pressing reset button, which restored "waiting for connection" message and restored pairing procedure, but old v4 menu stil is present. And no chance to flash v5.
When card with v5 inserted, after reboot yellow led is blinking for 30sec like during normal flashing process. But after that, still the same - v4 menu. 
Any idea how to remove v4 and revert back to factory or properly force v4 to v5 update?

TheCrypt0, 2022-09-04:
Hi,
Unlocking RTSP has been used by a lot of people and they correctly received their license in 24/48h to use the feature. If you donated and had any issue with the video stream please let me know and I'll be glad to help you!

There's no permanent yi-hack-v4, it's a normal flash procedure like other firmwares, since it appears that the web UI is kind of _inactive_ have you tired to force refresh the browser and clearing the cache?

Let me know if this helps,
Crypto

g1za, 2022-12-08:
Stock firmware can be found here, just in case:
https://github.com/shadow-1/yi-hack-v3#going-back-to-stock-firmware

If I correctly remember I just uploaded the files onto the card when I upgraded from v4 to v5

ses1er, 2023-01-09:
I've had the same issue trying to go from v4 to v5 with the existing ex-FAT formatted SD card. I ended using a FAT32 formatted 16GB SD card and the flash was successful. Please give that a try.