TheCrypt0/yi-hack-v4 #366: Help with identifying the camera

NiiiYaa, 2022-09-16:
Help with identifying the camera
I wish to apply the right 'hack' so I need to identify the camera.
The camera was bought on AliExpress - "no brand".
Works with the "YI IOT" app.
On GoogleHome identifies itself as: "xiaoyi".
Firmware: 6.0.05. 10_202112100938
ID: "A1769004DDT2Q6220729"

Motherboard:
Looks like this:
[image]

Chip 1:
ANYKA
AK3918EN080
V200
CDSJ21F22

Chip 2:
SV6256P
TAC2204
IR498B5
* I might be 1
* B might be 8

Chip3:
OK291323
ULN2803VS
* 8 might be B
Chip 2
ANYKA
AK3918EN080
V200
CDSJ21F22

Camera looks like this:
[image]

kalel681, 2022-10-30:
I also have this camera and am wondering if the Dome hacks will work?

jacobwhite, 2023-12-09:
Yall ever find out?

09Memfis, 2025-03-29:
Hi everyone!

I'm also trying to recover a dead Chinese IP camera using the **D011AP_MB_V1.6** board and **Anyka AK3918EJ08** SoC. My camera was working with the **YI IOT** app.

Symptoms:
- Red IR LEDs flash very dimly on power-up
- No blue LED or beeper sound
- Camera doesn't show up on the network
- I tried several 5V power sources – same result

I have logs and full photos of the board, and I'm currently trying to flash via microSD.

If anyone has a working firmware (`update.bin` or `uImage`/`romfs.img`/`rootfs.img`) or a dump from a similar camera, please share 🙏

Let me know if I can help test anything.

Thanks a lot!